ClubCedille/jardiniot #13: Attributs de qualité

BinaryTiger, 2016-09-29:
Quel seront les attributs de qualité visé et sur lesquels on peut faire des compromis?

(sécurité, fiabilité, utilisabilité, portabilité, performance, extensibilité, robustesse, etc)

mikefaille, 2016-10-06:
+ 
utilisabilité : accessible aussi  rapidement que possible pour mr et madame tout le monde
robustesse : éviter les dégâts physiques parce que ça peut en causer des considérables
maintenabilité : que ça se répare facilement. Que les pièces soient peu dispendieuses.
extensibilité : possibilité d'y ajouter des addons. ex : permettre d'autres formats. Avoir un serveur pour plusieurs "place où mettre la terre".

- (moins important)
sécurité : communication par câble. Peu de données confidentielles....
performance : peut de ressource sont utilisés. Pour ce qui est du seuil de performance, je parlerais plutot de fiabilité.

Pour ce qui est de la fiabilité, en fait, ça fait parti du principe que si le système n'est pas fiable si ce n'est que pour produire de la nourriture que ça va être un échec assuré. Avec la robustesse, s'assurer que les composantes sont solides, remplaçables et que la construction soit simple pour faciliter la reproduction. Alors, éviter les pièces personnalisés si possible.

cmccauley, 2016-10-06:
@mikefaille https://github.com/ClubCedille/jardiniot/wiki/Attributs-de-qualit%C3%A9

mikefaille, 2016-10-06:
@cmccauley merci ! (ton lien me redirige là. https://github.com/ClubCedille/jardiniot/wiki quel serait l'original ? ty)

cmccauley, 2016-10-06:
@mikefaille Je vais présumer que tu n'as pas le write access dans le Wiki. Mon link redirige vers Create new Page.

mikefaille, 2016-10-06:
Effectivement.

mikefaille, 2016-10-08:
@cmccauley  J'viens de mettre à jour ça. https://github.com/ClubCedille/jardiniot/wiki/Attributs-de-qualit%C3%A9
Ça pourrait être plus détaillé. De plus, je crois que bien que j'ai fait le travail en aval qu'il faudra recentrer le travail sur nos besoins.

AXDOOMER, 2017-05-12:
@mikefaille @BinaryTiger Est-ce que cette section est terminée?

AXDOOMER, 2017-06-10:
Fermé comme il n'y a pas eu de réponse après un mois.

closed 2017-06-10 by AXDOOMER

mikefaille, 2017-06-11:
ah, je n'ai pas vue. Il faut simplement garder en tête de s'assurer de remplir nos objectifs de design.

AXDOOMER, 2017-06-11:
Je pense que c'est compliqué. La question revient toujours à "C'est où qu'on arrête avec les features". 

#### Exemple de projets:

**Pot de fleur:** Projet très simple. La chaleur est pas fournie par le pot (soleil) et l'arrosage est fait manuellement. Suivi à l'oeil. 

**JardinIOT:** Pot avec alimentation automatique de la plante en nutriment et en haut. Chaleur et photosynthèse fournie par les DELs. Suivi fait par des capteurs. 

**Serre de SerreÉTS:** Méga système qui fait tout pour la plante, entièrement automatisé. Le suivi est fait par des systèmes à la fine pointe de la technologie. 

Tout dépendant des projets, certains objectifs de conception sont plus dures à respecter. Par exemple, le troisième système que j'ai mentionné peut causer des plus grands dégâts d'eau car il contient plein de tuyaux. Le système 1 est plus complexe que le système 2 et le 2 plus complexe que le système 3, un est moins robuste car il a plus de pièces que d'autres qui le constitue et qui peuvent lâcher. On veut fournir un système qui fait beaucoup, mais à un faible prix. C'est plus difficile de réparer un système plus complexe, donc il est moins maintenable. Le système qui fait plus est plus difficile à utiliser et à monter par les gens normaux. 

Parfois je me dis qu'on devrait arrêter de faire notre propre jardin comme maintenant on a un partenariat avec SerreÉTS et que eux sont bons pour en faire. Ça nous sauverais de la job (en même temps ça nous priverait d'un accomplissement) , mais on travaillerait dans ce qu'on est bon et le problème du bucket est laissé aux autres qui le régleront très bien. 

Comme c'est une plante par bucket, notre système revient déjà très cher. Pour le prix, on peut oublier ça.  C'est bon pour l'espace, certes on ne peut pas cultiver beaucoup de plante. Surtout que là va falloir en faire un beaucoup plus grand comme j'ai dessiné dans un schéma dans un autre issue. On est fier de notre bucket, mais je me demande si ça vaut la peine qu'on le continue. On en a fait beaucoup, mais ça doit pas être quelque chose qui nous empêche de prendre la décision qu'il faut concentrer nos efforts ailleurs. Par contre, c'est certain qu'on doit continuer à faire du IOT, peut-être pas avec ce projet là. Le système d'agglomération de données qu'on fait par contre, c'est gold. 

Il y aura plusieurs choses à discuter. J'ai mentionné qu'on devait faire une réunion entre les gens de jardinIOT dans le channel sur rocket chat, c'est un occasion de discuter et de prendre des décisions. Le statu quo n'est pas tout le temps bon et peut-être qu'en parlant, on verra si c'est une bonne idée et si les autres membres veulent, car je sais qu'il y en a qui tiennent beaucoup à ce qu'on fasse notre propre bucket, ce qui est tout à fait légitime. Mais c'est important de se remettre en question et d'en discuter pour renforcir nos convictions. Faut qu'on se confirme que la direction dans laquelle on va est la bonne et cela nous motiverait aussi d'avantage.

mikefaille, 2017-06-11:
Belle analyse.