ClubCedille/jardiniot #6: Options de languages / framework

BinaryTiger, 2016-09-29:
Pour la programmation du serveur et client. Pros/Cons. Fera l'objet d'un debat pour choisir les options finales.

AXDOOMER, 2016-10-04:
Mardi matin je vais surement venir dans le local de Cedille et commencer à faire de la recherche là dessus si je n'ai pas commencé en fin de semaine. N'hésitez pas à me faire des suggestions si pour améliorer ce que je fais.

AXDOOMER, 2017-01-21:
Comment ça fonctionne présentement:

On a un Arduino et un serveur Python (site web). On utilise MQTT pour faire la communication entre les deux. Le serveur affiche et permet de changer nos paramètres. Le Arduino est connecté aux senseurs et les autres trucs (lumières, ventilateurs) et reçoit nos paramètres. Il prend des décision en fonction de ceux-ci. Par exemple, il augmente la température si la température est plus basse que celle paramétrée. 

Ventilateurs, Lumières, Senseurs <--> Arduino <--> MQTT <--> Site web en Python

closed 2017-05-12 by AXDOOMER