ClubCedille/jardiniot #7: Planification des itérations

BinaryTiger, 2016-09-29:
Comment sera diviser la charge de travail. On peut s'inspirer des milestones qui sont faites ou bien faire autrement et adapter les milestones par la suite. Assez détaillé pour qu'on puisse mettre des dates et des objectifs.

mikefaille, 2016-10-01:
Je propose d'y aller par sprint de 2 semaines avec des tâches courtes qui se font en moyenne en 2-4 heures. Pour la liste des tâches, ça devient un backlog pour le kanban. Alors, tu accumules les tâches notés pour en avoir les autres semaines.

AXDOOMER, 2017-05-12:
> Assez détaillé pour qu'on puisse mettre des dates et des objectifs.

Avec la façon dont le projet avance présentement c'est impossible. On le fait de façon incrémentale et libre. Tout le monde choisi ce qu'il veut faire et travaille quand il veut sans obligation. Dans les workshops, on en fait le plus possible.

mikefaille, 2017-05-12:
@AXDOOMER C'est exactement ça le kanban. Le 2 semaines, c'est juste pour la review. Bien sur, le but c'est d'avoir aussi un livrable minimum viable après chacune des itérations.

AXDOOMER, 2017-05-12:
Personne ne se sert du Kanban présentement et il est totalement abandonné depuis des mois. Les étapes du projet projet sont déterminés par ma personne, comme par exemple: Le moment où le projet sort du stade de prototypage. Ce n'est pas écrit nulle part, mais j'en ai discuté avec Gotam et les autres membres du projet.

mikefaille, 2017-05-12:
Le Kanban c'est une méthode de gestion de projet par flow avant d'être une grille avec des lanes. 
Voir ici :
https://en.wikipedia.org/wiki/Kanban

mikefaille, 2017-05-12:
> Ce n'est pas écrit nulle part, mais j'en ai discuté avec Gotam et les autres membres du projet.

Être lean c'est le plus important. En autant que la communication soit efficace tant pour les membres actuel est les nouveaux.

AXDOOMER, 2017-05-12:
> Kanban

À quoi ça sert quand personne s'en sert? Faut pas trop perdre de temps à gérer le monde et à créer des issues pour dire ce qu'il y a à faire. Le projet avance de façon naturelle. 

Les gens font ce qu'ils veulent et sortent des cadres établis. De toutes les technologies qu'on pensait utiliser initialement, aucune n'a été utilisé dans le projet sauf le Arduino. La majorité de la documentation du wiki est à refaire. Même si on la fait là, ça risque encore de changer. 

Y'a beaucoup de choses qui ont été planifiées initialement et cette planification est rapidement morte, car le projet est très évolutif et a changé souvent. 

Présentement, je sais que le projet va sortir du stade de prototypage quand le serveur de Aubert va être finit, quand la communication entre les composantes va être finit et que tous les censeurs vont être plugués. Après, ça dépend ce que les contributeur vont vouloir faire. Ça devrait être la pompe à eau, mais peut-être que d'ici à quand ça va être finit, ceux-ci vont avoir d'autres idées.

mikefaille, 2017-05-12:
@AXDOOMER Qui a dit que c'était du Kanban avant ? Le board ?

AXDOOMER, 2017-05-12:
C'est qui/quoi le "board"?

mikefaille, 2017-05-12:
Ça : https://github.com/ClubCedille/jardiniot/projects/1

AXDOOMER, 2017-05-12:
Ça s'appelle "Main Kanban".

mikefaille, 2017-05-12:
Ce n'est pas une gestion de projet qui reflète la philosophie Kanban dans un cadre Agile. Toi tu en fais.

AXDOOMER, 2017-05-12:
Pour les itérations:

1. Sortir du stade de prototypage (requirements déjà établis)
2. Faire la pompe à eau et le filtrage
3. ???

Le site web dépend de l'API qui doit être finit. La structure de l'API existe et a été [documentée](https://github.com/ClubCedille/jardiniot/blob/master/serveur-web/README.md) par Aubert sur le wiki. Il peut être fait n'importe quand, surtout que le côté interface a [déjà été designé](https://github.com/ClubCedille/jardiniot/wiki/Interface-web).

AXDOOMER, 2017-05-12:
Quand je disais "à quoi sert le Kanban", je demandais pas c'était quoi, je voulais dire qu'il sert à rien.

AXDOOMER, 2017-05-12:
Je pourrais carrément effacer le Kanban et personne ne remarquerait. Même chose pour certaines parties du wiki, mais Aubert et moi on s'occupe de faire quelques mises à jour présentement.

mikefaille, 2017-05-12:
Toutefois, il faut donner le projet à notre ami @BinaryTiger. C'est grâce à lui que le projet s'est doté d'une vision. Certes, tu peux la changer. Ultimement, un projet sans vision, c'est un projet mort.

@BinaryTiger L'a parti comme un chef. Pour les pièces, nous avions besoins des donnés pour faire une projection du budget.  
Pour la mission, nous avions besoin du texte pour créer un demande de budget.

Tu peux te comparer plusieurs fois aux autres. Par contre, ça ne sert à rien de mettre en valeur tes bons coups si tu prends le temps pour dénigrer publiquement celui des autres par comparaison. Ça diminue totalement la qualité de ton travail si seulement il se veut social.

aubguillemette, 2017-05-12:
@mikefaille @AXDOOMER Je crois qu'encore une fois, vous vous êtes mal compris. Cette discussion n'a pas nécessairement sa place sur GitHub, je vous invie à suivre ce type de discussion sur RocketChat pour ne pas polluer les issues. Toutefois, pour cette conversation-ci, je vous invite à mettre les pendules à l'heure IRL puisque ce canal de communication semble faire en sorte que il y a beaucoup de 'miscomprehension'.

mikefaille, 2017-05-12:
@aubguillemette Je suis d'accord avec toi que ça n'a pas sa place. Je suis juste déçu lorsque l'on dénigre le travail des autres publiquement. Il me semble que l'on doit focuser sur ce que l'on réussi. Ça ne vaut pas la peine de regarder dans le passé surtout s'il n'est pas bien compris.

Props à @BinaryTiger

aubguillemette, 2017-05-12:
Voir le channel RocketChat du projet pour mon statement sur la situation.

AXDOOMER, 2017-05-12:
> Pour les pièces, nous avions besoins des donnés pour faire une projection du budget.

Pour référence: https://github.com/ClubCedille/jardiniot/wiki/Budget

> prends le temps pour dénigrer publiquement

> dénigre le travail des autres publiquement

C'est une question de perception et tu vois toutes mes idées comme une attaque. Pourtant mes commentaires se résument seulement à dire que le Kanban ne servait plus. Antoine a tout le mérite d'avoir créé le projet et d'avoir eu une vision pour nous convaincre d'embarquer. Où est-ce que j'ai dit le contraire? Je suis franchement tanné que tu me portes des intentions. "No hate" comme tu dis. C'est fou comment je peux être d'accord avec toi sur plein de choses, mais que tu penses que je pense le contraire.

mikefaille, 2017-05-12:
Ce que je propose comme perspective c'est : `mieux comprendre le kanban ne vaut pas de discréditer le travail d'un autre.`
Genre, au moins dit : 
- y'a eu un excellent travail. Je reconnais XYZ de bons
- y'a ABC que j'aime moins pour tels raisons. J'accepte néanmoins que notre amis devait avoir une bonne raison d'avoir cette approche. Pour l'instant, je ne comprends pas.
- Avec le temps dû «i», je crois que j'ai réussi à obtenir DEF comme évolution à ABC tout en soutenant XYZ. Je crois que l'on est sur la bonne vois.

aubguillemette, 2017-05-12:
@AXDOOMER @mikefaille Je vous ré-invite à avoir cette conversation ailleurs que sur ce thread. Je réitère: je ne crois pas que cette conversation devrait avoir lieu sur GitHub.

mikefaille, 2017-05-12:
@AXDOOMER On s'est entendu en privé.

AXDOOMER, 2017-05-12:
@mikefaille  Je peux pas être plus d'accord avec toi que ça sur ta perspective. Je n'ai pas mis plein de commentaires positifs partout, mais ça veut pas dire que mon but est de basher le travail d'un autre. Si je mets pas de commentaires positifs et que je dis de quoi de négatif sur quelque chose, il ne faut pas assumer que tout le reste n'est pas bon. 

Je vais dire ceci dans le but de rendre mes intentions claires: Je trouve que le travaille d'Antoine était bon et je ne l'ai pas mentionné car je crois que nous sommes tous d'accord (incluant ceux qui ne font pas partie de cette discussion). Aujourd'hui on a construit sur le sien et ce qu'il avait apporté au projet au début a évolué en de quoi d'encore plus meilleur. S'il n'avait pas été là au début, le projet ne serait rien aujourd'hui. Un peu pour reprendre tes mots, mais en l’appliquant à cette situation, le projet serait mort sans sa vision.

mikefaille, 2017-05-12:
Merci de comprendre !

mikefaille, 2017-05-12:
De mon côté, il est entendu que le mot dénigré était trop fort et que discrédité était mieux adapté. Je n'avais pas réalisé à quel point sur le champs. J'ai alors fait les changements nécessaire tel que discuté.

closed 2017-05-12 by mikefaille