ClubCedille/jardiniot #8: Budget

BinaryTiger, 2016-09-29:
High / median / low estimates. Selon les options proposées dans la section des composantes. Estimation pour le matériel de construction du bucket.

AXDOOMER, 2017-05-12:
Ce n'est plus important comme tout a été acheté. Si jamais on veut refaire le projet, un estimation des coûts se trouve  [sur le wiki](https://github.com/ClubCedille/jardiniot/wiki/Budget).

closed 2017-05-12 by AXDOOMER

mikefaille, 2017-05-12:
En effet, c'est lui qui a monté le wiki. Je crois que la tâche n'a juste pas été fermé sur le moment.

AXDOOMER, 2017-05-12:
La page du budget n'est pas à jour non plus. Il y a plusieurs composantes affichées et calculées dans le prix du projet qui ne sont pas utilisées. @cmccauley Peux-tu te charger d'enlever les composantes non utilisées de la liste comme c'est toi qui t'es chargé de les acheter? Merci.