ClubCedille/jardiniot #14: Authentification entre le controlleur et le device

BinaryTiger, 2016-11-02:
Pairing entre le controlleur du bucket le dispositif de l'utilisateur a l'aide d'un [3 digit led display](digikey.com/product-search/en?mpart=LTC-4724JF&vendor=160). 

Le controleur va generer 3 chiffres au hasard qui seront afficher sur le display. A l'aide de son mobile ou de son ordinateur, l'utilisateur pourra faire un pairing et configurer le jardin pour la premiere fois (wifi, etc). Le display devrais etre facile a connecter et a deconnecter, laissant ainsi l'utilisateur la chance d'utiliser le meme display pour configurer tout ses jardins. En bonus, si le display n'est pas connecter, c'est une protection de plus.

Chose a faire: 
* design de ce module a haut niveau
* pseudo code de comment le module devrait gerer l'affichage et le pairing

BinaryTiger, 2016-11-02:
Si vous avez d'autre idee, elles sont les bienvenues. Ajoutez votre idee dans les commentaires du issue.

AXDOOMER, 2017-05-12:
Cette situation a été abordée par: https://github.com/ClubCedille/jardiniot/issues/22 (Wi-Fi Manager)

Les autres cas sont présentement réglés ailleurs.

closed 2017-05-12 by AXDOOMER