ClubCedille/jardiniot #15: Diagramme de composantes

BinaryTiger, 2016-11-02:


AXDOOMER, 2017-05-12:
Le circuit électrique a été fait par @cmccauley. Il sera documenté sur le Wiki.

AXDOOMER, 2017-05-12:
L'autre diagramme a été fait par Jananjoy. Il a pas l'air à avoir un compte GitHub et c'est @ygotame qui a ajouté le schéma au Wiki. [Schémat/diagramme ici](https://github.com/ClubCedille/jardiniot/wiki/schema)

Il doit être mis à jour, car c'est pas comme ça que fonctionne le projet.

AXDOOMER, 2017-05-12:
Ça serait bien d'**avoir le code source** du schéma pour pouvoir le modifier dans le futur.

AXDOOMER, 2017-05-12:
On a maintenant les photos de la board et son circuit électrique fait par Charles. Merci à @aubguillemette  pour les photos. 

https://github.com/ClubCedille/jardiniot/wiki/Board-et-ses-composantes  <-- Ne pas cliquer sur ce lien si vous êtes préoccupés par votre bande passante.

closed 2017-07-08 by AXDOOMER