ClubCedille/jardiniot #19: mqtt - Créer les topics  "Temperature" et "Humidite"

mikefaille, 2017-03-06:
Pour mqtt. moquer des tests avec la "Temperature" et l'"Humidite"
S'assurer d'utiliser des valeurs en int comme format de transmission/réception (éviter le format json).

Dans le serveur web, moquer l'envoie des données pour fin de test sans le esp (facilitant le développement..).

Les données devront être dans un format clef valeur (topic/value).

AXDOOMER, 2017-03-06:
Correction: La topic est un pot et plusieurs données relatives au pot sont envoyés dans une topic.

mikefaille, 2017-03-06:
Voici un example pour les bests pratices : http://www.hivemq.com/blog/mqtt-essentials-part-5-mqtt-topics-best-practices
Je suis toujours d'avoir que ça devrait être une clef en fonction de ma référence. C'est aussi comme ça que c'est designé.

AXDOOMER, 2017-03-06:
Je crois l'avoir fait de la sorte car ma topic contient les données sous forme de Jason, donc il y en a plusieurs. Comprendre les meilleures pratiques ce sera un objectif pour quand ont entreprendra le travail là-dessus. Par contre, si créer plein de topics devient trop compliqué sur le ESP, le Jason reste un option. 

http://www.hivemq.com/blog/mqtt-essentials-part-5-mqtt-topics-best-practices
@aubguillemette

mikefaille, 2017-03-06:
> Comprendre les meilleures pratiques ce sera un objectif pour quand ont entreprendra le travail là-dessus. 

Ça devrait bien aller avec plusieurs topics : 

jardinIOT/sensors/temprature
jardinIOT/sensors/humidite
jardinIOT/commands/leds

AXDOOMER, 2017-03-06:
> jardinIOT/commands/leds

DEL en français.

aubguillemette, 2017-03-06:
Je crois qu'en effet, la meilleure pratique, ce serait d'utiliser un système de clé-valeur, autrement dit, chaque sensor devrait avoir son topic. C'est comme ça que je le voyais au début anyway.

aubguillemette, 2017-05-12:
Voir https://github.com/ClubCedille/jardiniot/wiki/Connectivit%C3%A9-entre-ESP-et-API-(MQTT)

closed 2017-05-12 by aubguillemette