ClubCedille/jardiniot #21: Compléter le système embarqué

AXDOOMER, 2017-03-06:
#### Maintenant qu'on peut envoyer des données du DHT au serveur MQTT, la prochaine étape est la suivante: 

(Le travail s’effectuera ici: https://github.com/AXDOOMER/jardiniot et résultera en un PR)

### À faire (durant le développement)
- [ ] Permettre au ESP de recevoir des topics (il peut déjà en envoyer)
    - Un pot a plusieurs topics qui lui sont attachées.
- [ ] Implémenter toutes les bonnes topics (consultez https://github.com/ClubCedille/jardiniot/issues/19 comme référence)
    - Humidité, température, contrôle des ventilateurs, contrôle des lumières DEL et autres.

### À faire après le développement (avant le PR)
- [ ] Améliorer les commentaires
- [ ] Enlever le code mort
- [ ] Enlever tout ce qui est inutile (évidemment!)

mikefaille, 2017-03-07:
Je te propose de faire un pull request par tâche autrement ça l'implique que tu vas travailler tout seul car en principe, un pull request sert à gérer le QA pendant le développement et non après.

AXDOOMER, 2017-04-17:
### Pour recevoir des topics

Exemples de code:
* http://www.esp8266.com/viewtopic.php?f=29&t=2745
* https://www.baldengineer.com/multiple-mqtt-topics-pubsubclient.html

AXDOOMER, 2017-04-18:
#### Pour contrôler les lumières

```
void setup() {
  // put your setup code here, to run once:
  pinMode(5, OUTPUT); 
  setPwmFrequency(5, 1);
  pinMode(6, OUTPUT); 
  setPwmFrequency(6, 1);
}

void loop() {
  analogWrite(5, 255);
  analogWrite(6, 255);

}

void setPwmFrequency(int pin, int divisor) {
  byte mode;
  if(pin == 5 || pin == 6 || pin == 9 || pin == 10) {
    switch(divisor) {
      case 1: mode = 0x01; break;
      case 8: mode = 0x02; break;
      case 64: mode = 0x03; break;
      case 256: mode = 0x04; break;
      case 1024: mode = 0x05; break;
      default: return;
    }
    if(pin == 5 || pin == 6) {
      TCCR0B = TCCR0B & 0b11111000 | mode;
    } else {
      TCCR1B = TCCR1B & 0b11111000 | mode;
    }
  } else if(pin == 3 || pin == 11) {
    switch(divisor) {
      case 1: mode = 0x01; break;
      case 8: mode = 0x02; break;
      case 32: mode = 0x03; break;
      case 64: mode = 0x04; break;
      case 128: mode = 0x05; break;
      case 256: mode = 0x06; break;
      case 1024: mode = 0x07; break;
      default: return;
    }
    TCCR2B = TCCR2B & 0b11111000 | mode;
  }
}
```

closed 2017-05-20 by AXDOOMER