ClubCedille/jardiniot #23: Implémentation d'un serveur PostgreSQL centralisé

aubguillemette, 2017-03-20:
@xavsavage et moi-même sommes en train d'implémenter le côté site web du projet et nous nous servons d'une base de données PostgreSQL. Serait-il possible d'héberger une base de données PostgreSQL sur un des serveurs de Cédille? De cette manière, nous pourrions nous servir de la même base de données et cela simplifierait notre processus de développement.

De plus, si l'on héberge le site de JardinIoT sur l'un des serveurs de Cédille, on pourra se resservir de la même base de données sans avoir besoin de la réimplémenter.

Merci en avance de l'attention que vous porterez à cette requête.

mikefaille, 2017-03-20:
@aubguillemette purema4 a terminé le design pour profiter du stockage ZFS sur notre environnement serveur de production. Il ne reste plus qu'à l'implémenté cette semaine.

Pour des fins de développement, je te propose de générer des données bidons afin de toujours travailler avec les mêmes données. 

- Pour la génération des données, ça ne peut que ce faire qu'à un seul endroit. 
- Pour l'acquisition des données,  c'est là qu'il est possible de prendre des données déjà collectés.

Autrement-dit, ma proposition pour développer votre solution, c'est de faire du mocking de votre système.
https://fr.wikipedia.org/wiki/Mock_(programmation_orient%C3%A9e_objet)

AXDOOMER, 2017-03-29:
Gotam et Aubert m'ont dit que ça été faire aujourd'hui, par contre il faut document la procédure d'installation détaillée pour qu'un usager sans expérience puisse le faire. 

https://github.com/ClubCedille/jardiniot/wiki/Installation-du-serveur-web

closed 2017-05-05 by aubguillemette