ClubCedille/jardiniot #31: Faire le site/le control panel

aubguillemette, 2017-05-07:
Commencer à faire le site qui va servir de control panel pour le jardin.

AXDOOMER, 2017-05-12:
Voir [le design du site web](https://github.com/ClubCedille/jardiniot/wiki/Interface-web).

AXDOOMER, 2017-05-12:
Alexandre Brochu était peut-être intéressé à contribuer aussi, par contre son username est pas disponible pour le taguer ou l'assigner.

AXDOOMER, 2017-05-12:
Trouvé! @AlexBrochu

AXDOOMER, 2017-06-06:
### Important

Nous travaillons maintenant avec  [SerreÉTS](http://serreets.com/). Notre système de site web et serveur qui recueillent les données doit être extensible et général. 

Il doit marcher avec n'importe quel senseur, que ce soit pour recueillir des données ou controller un bidule (comme un pompe ou une lumière). 

**On doit pouvoir stoker les données des différents senseurs en indiquant une marge d'erreur**, ceci afin qu'il soit possible d'utiliser les données pour des recherches scientifiques.

notarock, 2018-06-21:
Le site est dans la version master :+1: 
Pour le moment, Il permet d'afficher les données en temps réel à interval de 1 seconde.

[video](
https://www.youtube.com/watch?v=bx7-ICqu1vo)

AXDOOMER, 2018-06-22:
Fermé en faveur de https://github.com/ClubCedille/jardiniot/issues/48

closed 2018-06-22 by AXDOOMER