ClubCedille/jardiniot #33: [API] Modification du connecteur pour l'envoi de commandes au ESP

aubguillemette, 2017-05-09:
- [ ] Renommer connector.js en mqtt_connector.js (ce sera plus évident de distinguer ce que ce fichier fait)
- [ ] Convertir le connecteur en module node.js pour qu'il puisse être utilisé comme un module (voir [sqlite_connector.js](https://github.com/ClubCedille/jardiniot/blob/master/serveur-web/api/sqlite_connector.js) à titre d'exemple.
- [ ] Ajouter le code pour envoyer des commandes au ESP en se basant sur les spécifications décrites [ici](https://github.com/ClubCedille/jardiniot/wiki/Connectivit%C3%A9-entre-ESP-et-API-(MQTT)).

aubguillemette, 2017-05-09:
- [ ] Ajouter du code pour pouvoir "Register" un nouveau bucket dans la base de données (via une requête HTTP POST)

- [ ] Ajouter du code pour envoyer des commandes au ESP (via MQTT, voir le code du connecteur)

AXDOOMER, 2018-06-22:
@notarock  Je ferme ce issue. Je crois qu'il n'est plus nécessaire, car je crois que tu vas faire ça différemment.

AXDOOMER, 2018-06-22:
Je vais effacer cette branche aussi. Définitivement trop vieille et abandonnée. 

https://github.com/ClubCedille/jardiniot/tree/feat_connecteur_SubscriberTopicsAuto

closed 2018-06-22 by AXDOOMER