ClubCedille/jardiniot #36: [API] [ESP] Question des délais et de l'envoi de commandes

aubguillemette, 2017-05-12:
@AXDOOMER et moi avons discuté de la possibilité d'envoyer le status des sensors du bucket à une plus grande intervalle. Nous avons aussi énoncé la possibilité d'envoyer les commandes de l'API au ESP une fois par minute.

Cependant, la manière dont j'avais imaginé l'API ne conservait pas les commandes envoyées dans la base de données. Nous avons donc énoncé l'idée que le ESP pourrait garder la dernière commande envoyée en mémoire et la renvoyer à chaque minute au Arduino pour assurer l'intégrité de la commande.

J'aimerais avoir votre opinion là-dessus. Il est déjà convenu (selon #35) que l'on assure la qualité des données en empêchant la corruption de celles-ci. Est-ce qu'on devrait assurer la qualité des commandes reçues par l'ESP aussi (en empêchant la corruption de ces commandes) ou devrait-t-n assurer la qualité des commandes reçues par l'ESP en envoyant à intervalle donnée la même commande?

aubguillemette, 2017-05-12:
La raison pour laquelle je suis intéressé à savoir ce que vous en pensez, c'est que cette décision va influencer le code de l'[API](../tree/master/serveur-web)

AXDOOMER, 2017-05-12:
@Knorax

Knorax, 2017-05-12:
Je penses qu'on pourrait appliquer le hamming code autant dans l'envoi des status de sensor arduino vers l'ESP que dans l'envoi des commande de l'ESP vers l'arduino, on aurait pas besoin d'envoyer la dernière commande continuellement comme ça.

Juste pour être certain tho, ça ressemblerait à ça right ? :
Sensor->arduino(applique le hamming code)-> ESP(reçois le hamming et l'interprete)->data base->api(pour setter la temperature/humidité)->ESP(recois la temperature/humidité voulu et converti en commande)->arduino->systeme du bucket

aubguillemette, 2017-05-12:
Drette ça, ça ressemblerait à ça oui.

aubguillemette, 2017-05-12:
J'suis d'accord avec ça moi. Ça éviterait de changer le code/la conception de l'API pis on assurerait la qualité des données.

Knorax, 2017-05-12:
pis les protocole d'envoi de donnée par wifi avait pas un checksum pour s'assurer que ça soit gucci à la reception ? (Log100 flash backs)

aubguillemette, 2017-05-12:
Y'est sensé avoir de la vérification de données au niveau du protocole (vu qu'on communique via MQTT sur TCP)

aubguillemette, 2017-05-12:
Although si l'ESP ou l'API se chie dans son envoi, le protocole va s'assurer que la chiure qu'on envoie est la chiure reçue.

Knorax, 2017-05-12:
Tant que ce que l'API envoi est pareil que ce que l'ESP reçoit moi je crois qu'on est noice avec ça. Si il y a des chiure, on va savoir que c'est nous :P

aubguillemette, 2017-05-12:
Comme on dit dans le métier, un code 8! ;)

Seems good. On va y aller comme ça.

mikefaille, 2017-05-12:
C'est l'code 18 champion --^
https://www.urbandictionary.com/define.php?term=Code%2018