ClubCedille/jardiniot #40: [GÉNÉRAL] Choisir une license pour le projet

aubguillemette, 2017-05-12:
- [ ] Choisir une license pour le projet
- [ ] Créer un fichier LICENSE dans le repo
- [ ] Ajouter la license dans les fichiers package.json

Ci-dessous, j'ai mis un tableau que j'ai pris sur le [blog de Jeff Atwood](https://blog.codinghorror.com), dans l'article (Pick a License, Any License)[https://blog.codinghorror.com/pick-a-license-any-license/].

License|Source|Type (Clauses)|Description
--------|--------|--------|--------
|None|Open|None (0)|Without a license, the code is copyrighted by default. People can read the code, but they have no legal right to use it. To use the code, you must contact the author directly and ask permission.|
|Public domain|Open|Permissive (0)|If your code is in the public domain, anyone may use your code for any purpose whatsoever. Nothing is in the public domain by default; you have to explicitly put your work in the public domain if you want it there. Otherwise, you must be dead a long time before your work reverts to the public domain.|
|GPL|Open|Copyleft (12)|The archetypal bearded, sandal-clad free software license. Your code can never be used in any proprietary program, ever! Take that, capitalism!|
|LGPL|Open|Mostly Copyleft (16)|GPL with a cleverly-constructed pressure valve release. Your free software can be binary linked to proprietary programs under certain very specific circumstances.|
|MIT/X11|Open|Permissive (2)|Short and sweet. Includes generic legal disclaimer of liability.|
|BSD|Open|Permissive (2)|Short and sweet. Includes legal disclaimer of liability with explicitly named organization.|
|Apache|Open|Permissive (9)|Requires derivative works to provide notification of any licensed or proprietary code in a common location.|
|Eclipse|Open|Permissive (7)|Business friendly. Allows derivative works to choose their own license for their contributions.|
|Mozilla|Open|Weak Copyleft (13)|Allows liberal mixing with proprietary software.|
|MS Permissive|Open|Permissive (3)|Resembles the MIT and BSD licenses. Not formally accepted by OSI, and also offered in a "Windows-only" LPL variant.|
|MS Community|Open|Copyleft (3)|Resembles the GPL license. Requires all contributed code to be returned to the community. Not formally accepted by OSI, and also offered in a "Windows-only" LCL version.|
|MS Reference|Proprietary|Read Only (3)|You can review the code, or make copies of it, but you can't use it or change it in any way. Allows a window (no pun intended) on formerly completely proprietary, secret code.|

JulienRobitaille, 2017-05-12:
Il y a aussi ce site: https://tldrlegal.com/ qui est super

aubguillemette, 2017-05-12:
Moi, j'opte pour BSD ou GPL.

BinaryTiger, 2017-05-12:
+1 pour [GPL3](https://tldrlegal.com/license/gnu-general-public-license-v3-(gpl-3))

AXDOOMER, 2017-05-12:
C'est quoi ça ce question là??? Choisir une license???? Y'aurait-il des hérétiques dans le club?? Cedille c'est un club de logiciels libres, pas un club de logiciels open source bidons à la noix. :boom:  :gun: 

[image]
**Choisi ta license. GPL3 ou GTFO.**

AXDOOMER, 2017-05-12:
Sur une touche plus sérieuse, faut faire certain de ne pas copier du code sur stackoverflow ou d'un autre auteur pour mettre ça sous GPL, sinon c'est du copyright infringement.

mikefaille, 2017-05-13:
Pour le `copyright infringement`, ça l'est aussi sans que ça soit GPLv3 en principe. Pour être stricte avec la loi, tout le code est prioritaire s'il n'a pas une licence déterminé.

AXDOOMER, 2017-05-14:
Dans ce cas là je trouve ça important, car la distribuer sous une license implique qu'on va mettre notre nom dans le copyright header.

AXDOOMER, 2017-06-09:
# VOTE
@aubguillemette @xavsavage @AlexBrochu @BinaryTiger @cmccauley 

*Je crois avoir tagué tous les membres du projet, mais au cas où j'en aurais oublié un, veuillez le taguer.*

Utilisez la réaction :+1: pour voter **OUI** pour la GPL3 ou :-1: pour voter **NON** sur ce commentaire. 

Si le vote **NON** gagne, nous discuterons d'un autre license. Le vote se porte sur la GPL3, car elle semble être la plus populaire dans ce issue. 

Le vote se termine le 23 juin ou dès que tous les gens tagués auront voté.

mikefaille, 2017-06-09:
GPL3 implying GPL3+?
(juste comme ça)

AXDOOMER, 2017-06-09:
J'ai modifié le vote, comme seulement toi et moi avons voté et nous semblons être d'accord. 

Ce sera donc GPL3+.

aubguillemette, 2017-06-09:
tfw @AXDOOMER call un vote à 2h du mat... ;)

AXDOOMER, 2017-06-30:
Je pense bien que c'est **GPL3+** qui a gagné. Le code source doit être mis sous cette license pour la sortie du prototype.

AXDOOMER, 2017-07-05:
Il ne reste plus qu'à ajouter les GPL3+ headers à nos fichiers source respectifs.

AXDOOMER, 2018-01-28:
Je ferme ça. C'est la responsabilité des membres d'ajouter leurs copyright headers au code qu'ils écrivent.

closed 2018-01-28 by AXDOOMER