ClubCedille/jardiniot #41: [ARDUINO] Faire le schéma électrique

aubguillemette, 2017-05-12:


AXDOOMER, 2017-05-12:
Quelque chose [comme ça](https://github.com/ClubCedille/jardiniot/blob/master/jardiniot-emb/MONTAGE.md)? Faut t'explique c'est quoi.

aubguillemette, 2017-05-12:
[Exemples de schémas électrique](https://www.google.ca/search?q=schema+électrique)

AXDOOMER, 2017-05-12:
Ok, c'est un truc à faire plus vers la fin. Qui se porte volontaire?

AXDOOMER, 2017-06-10:
@cmccauley Est-ce que tu as de la documentation sur la board? À date, on a des photos de la board pour savoir de quoi elle avait l'air si on la perd et j'ai pu documenter ces pins:
[image]
Par contre, je ne réussi pas à faire fonctionner le ventilateur.

notarock, 2018-03-20:
Si je me trompe pas, c'est ce que @CedrickGauthier fait en ce moment (C'est bien toi? Je croise les doigts pour ne pas déranger personne inutilement)

AXDOOMER, 2018-03-21:
Oui c'est lui.

AXDOOMER, 2018-07-05:
Mergé: https://github.com/ClubCedille/jardiniot/pull/61

AXDOOMER, 2018-07-05:
En ce moment, on dépend encore sur ça: https://github.com/ClubCedille/jardiniot/wiki/Composantes-%C3%A9lectroniques