ClubCedille/jardiniot #44: [DOCU] Comment faire pousser une plante?

AXDOOMER, 2017-06-06:
## Intro

Il faut trouver de la documentation pour savoir c'est quoi le meilleur environnement pour faire pousser une plante. Le club [SerreÉTS](http://serreets.com/) va nous aider en fournissant leurs connaissances. 

Ce qui nous avons fait présentement était au meilleur de nos connaissances, le prototype comprend des erreurs auxquelles nous devons apporter des corrections dans le futur. 

Ce post comprend les données recueillies jusqu'à date. 

## Faire germer des graines

Il faut mettre les graines dans ce qu'on appel un **moyen de germination**. Ça donne un endroit où la graine pourra faire sortir ses racines initiales. La graine est mis à l'intérieur, dans le milieu, du **moyen** qui est une genre de laine minérale. 
[image]
En fait, c'est un produit de noix de coco. Ça peut être acheté pour pas cher sur Amazon. 
[image]
Ça se peut qu'on doit mettre ça sur une plaque chauffante pour simuler un printemps pour faire que la graine germe en dedans de la mousse. Une fois les racines sorties, là les graines sont assez matures pour mettre ça dans un pot comme celui du jardin (faut pas sortir la graine du moyen, faut la laisser dedans). 

## Comment continuer à faire pousser la plante après

Si on veut pas tuer notre plante, il faut lui donner un environnement où elle aura les ressources nécessaires à sa survie. 

Présentement, il y a un genre de petit pôt de style grillage de plastique noir dans le fond du bucket. C'est là où on devra déposer notre graine germée. Elle devra tremper dans l'eau qui contient les minéraux afin de s'alimenter. L'eau contiendra des **billes d'argile** afin de simuler la terre. 

[image]

C'est plus facile pour des racines de pousser entre les billes que de pousser dans de la terre. On ajoutera des **nutriments** dans l'eau. On devra déterminer ce que ça sera. Il faudra en mettre assez mais pas trop (règle de trois) pour en avoir une bonne concentration. L'eau n'aurait pas besoin d'être filtrée, par contre **il faut quelque chose pour la remuer, sinon elle va stagner**. Afin de mesurer la quantité du nutriments, il faut utiliser une **sonde de conductivité**, qui pourra être fournie par SerreÉTS. 

Il faut aussi **garder un œil sur l'humidité, car s'il y en a trop, il va y avoir de la moisissure**. C'est très important. D'ailleurs, si on veut mesurer si on donne la bonne quantité de lumière à la plante, il faut utiliser une camera infra-rouge pour regarder quelle(s) sorte(s) d'ondes sont absorbées. Généralement, c'est rouge et bleu. Certaines graines requiert du orange pour simuler un certain moment de l'année, pour faire des fleurs. 

Une bonne manière de se faire un bucket serait la suivante, car présentement le bucket prototype est même pas assez gros pour faire pousser des tomates ou des piments. 

### PHOTO À VENIR
(Bukets - les mettre un à l'envers de l'autre, ça fait un super gros bucket très haut avec 2 buckets)

mikefaille, 2017-06-06:
Man, c'est tout que de l'art ! Belle documentation ! C'est peut-être fou mais des images rend un bunch de texte interminable plus agréable à lire.

_Ce qui nous avons fait présentement était au meilleur de nos connaissances, le prototype comprend des erreurs auxquelles nous devons apporter des corrections dans le futur._

^--- Real talk

AXDOOMER, 2017-06-22:
Une **poubelle** pourrait être un meilleur bucket, car c'est plus gros et c'est déjà monté. Il y a moins de travail à faire, car on a juste à scier un trou pour les DELs dans le couvercle et mettre un support pour ce qu'on va faire pousser dans le fond. 
[image]