ClubCedille/jardiniot #48: [FRONTEND] Conception du front-end

AXDOOMER, 2017-06-11:
### Pages

Il y a quatre pages conçues à date. Elles illustrent le fonctionnement de l’application web (site web). 

Page d'index quand on se connecte. On pourrait afficher de quoi de par défaut dans l'espace vide (ou même rendre ça personnalisable). Il faut utiliser le menu pour naviguer et ça marche sur téléphone mobile. Les autres pages pourraient être *Senseurs*, *Ajout senseurs*, *Ajout configuration senseur*, *Graphiques* (diagrammes pour interprétation), etc. Suggérez des idées. 
[image]

La page du statu des senseurs qui ont été ajoutés. 
[image]

L'ajout d'un senseur duquel on veut recueillir des données. Il faut faire ça pour qu'ils s'affichent. 
[image]

Avant de pouvoir ajouter un senseurs, il doit avoir sa configuration ajoutée au programme pour qu'il sache interpréter les données. 
[image]

### Raisonnement derrière cette conception

C'est le serveur qui enverra au Arduino des directions pour lire des senseurs. Le Arduino ne sait pas sur quelles pins sont pluguées les senseurs, on lui dit par l'interface web. Le site web contient des configurations par défaut, les mêmes sont programmées sur le Arduino. Certains senseurs ont besoin d'une library, d'autres non.

Enfin, ce qui ce passe, c'est que le Arduino recevra un message le disant qu'il y a un DHT-22 connecté sur sa pin 3 en entrée et 4 en sortie (exemple), avec un nouveau ID. Le Arduino ajoute ça à sa liste de senseurs connus et se met à transmettre les données qu'il lit. Le serveur voit des messages avec ce ID et les attributs au bons senseurs dans son interface. Le ID doit toujours être une nouveau ID (pas un ID recyclé), car la configuration peut être changée sur le serveur sans que le Arduino soit tout de suite mis-à-jour à cause du délais, alors dans le cas d'un ID recyclé, le serveur recevrait des infos ne venant pas du senseur qu'il pense, la mise-à-jour n'ayant pas encore été complétée. 

J'en suis venu à cette conclusion en parlant avec @mikefaille, on cherchait à concevoir de quoi de très modulaire où l'utilisateur n'aurait pas à modifier le code du Arduino ou du ESP, seulement configurer quelque chose par l'interface graphique.

AXDOOMER, 2017-06-16:
Faut pouvoir gérer la planification de tâches (exemple: utiliser la pompe pour vider l'eau à minuit puis reremplir). Faut aussi gérer les tâches qui se déclenche lors de certains événements (non planifiées), comme par exemple: l'humidité a montée à 80%, on démarre les ventilateurs.

AXDOOMER, 2017-06-20:
TODO: 

* Protocole de communication 
* Structure des classes (modèle du domaine)
* Pseudo-code Arduino
* Navigation du site web

AXDOOMER, 2017-06-20:
# Conception serveur et Arduino

[image]
[image]

AXDOOMER, 2017-06-20:
@JulienRobitaille La conception du front-end doit t'intéresser. On a besoin d'une personne de talent qui sera prête à la faire.

AXDOOMER, 2017-06-20:
La représentation d'un senseur se le Arduino devrait ressembler à ceci:
```
int id                    // # unique, reçu à partir du serveur. Doit toujours être unique. 
enum Type                 // DHT (1),  1 étant le numéro assigné par l'ordre dans le Enum. Il doit être le même sur le serveur. 
int FrequenceUpdate       // Fréquence à laquelle une donnée de ce senseur est recueillie puis envoyée au serveur.
vector<pinOutput>
vector<pinInput>
```

### CODE D'ALEX BROCHU À VENIR

JulienRobitaille, 2017-06-26:
@AXDOOMER Est ce que les maquettes faites par Antoine sont la direction que nous prenons ? Si oui, je pourrais les commencer.

AXDOOMER, 2017-06-26:
@JulienRobitaille  Il faudrait les revisiter pour voir si elles correspondent encore à nos besoins. D'après moi, elles doivent être modifiées, car elles sont vieilles et nos besoins ont changés. On pourrait se planifier une rencontre pour en discuter. 

Je suis dispo mardi en fin d'après-midi ou jeudi durant la journée ou en soirée.

AlexBrochu, 2017-09-28:
# Conception serveur et Arduino V2

[image]

Je propose une nouvelle architecture pour la conception serveur et Arduino.

aubguillemette, 2017-09-28:
Looks good to me! Beau travail!

<https://user-images.githubusercontent.com/10930804/30972210-ea111b1a-a437-11e7-8cc3-fb6d8fad2913.png>

notarock, 2017-11-04:
# Front end 2.0
### Menu Principal
Dès l'acces au site web, on se retrouve sur la page Index qui affiche tous les buckets connectés à l'api. 
[image]
Dans le header de la page se trouve un menu ou on peut acceder rapidement à la liste des buckets, la liste des configurations de senseurs et à certaines fonctionnalités comme ajouter un bucket ou une configuration de senseur.

### Buckets

Lorsqu'on clic sur un des buckets parmis la liste de l'index, on se retrouve sur la page qui affiche les informations sur ce bucket ainsis que les divers senseurs installés sur ce bucket.
[image]
Tout près du nom de bucket se trouve un bouton qui permet d'en faire la gestion. Depuis ce bouton, on peut  soit installer un nouveau senseur, ou supprimer le bucket (ce qui supprime la liste de senseurs associés). Lorsqu'on tente de supprimer le bucket, un message de confirmation s'affiche.
[image]
Lorsqu'on confirme la supression, le bucket et la collection est supprimé et on retourne à l'index des buckets.

Il est aussi possible d'ajouter un bucket avec ce formulaire accessible depuis la bar de navigation
[image]
Après l'ajout, on se retrouve sur la page de ce bucket pour pouvoir rapidement installer un senseur.

En sélectionnant l'option dans l'onglet de gestion du bucket (À droite du nom), on peut acceder au formulaire pour "Ajouter un senseur" sur le bucket. Il faudrait voir si on préfère utiliser "Installer un senseur" ou "Ajouter un senseur" pour être certain que la fonctionnalités soit distincte à l'ajout de configuration de senseurs.
[image]
Depuis la page de présentation du bucket, on peut ensuite sélectionner un senseur pour afficher les données qu'il recceuil.
[image]
Il est aussi possible de supprimer (Retirer du bucket) un senseur depuis cette page. Il y aura donc un message de confirmation semblable à celui de la suppression d'un bucket. Après avoir retirer le senseur, on se redirige sur la page du bucket duquel on vient de retirer le senseur.

### Senseurs

À partir du bouton dans la barre de navigation, on peut acceder à liste des configurations de senseurs. 
[image]
Il est aussi possible de créer des configurations de senseurs à partir de la barre de navigation. On accède donc à ce formulaire qui permet de spécifier le type de données receuillis par différents senseurs.
[image]
Lorsque la configuration a été créer, on est retourné sur l'index des configurations de senseurs.

Il est aussi possible d'afficher les informations d'une configuration de senseur à partir de la liste. En cliquant sur un des objets de la liste, on affiche donc son nom ainsis que les type de données receuillis par ce senseur.
[image]
Pour supprimer une configuration, il faut seulement l'afficher puis sélectionner l'option "Supprimer". Il y aura donc un message de confirmation semblable à celui de la suppression d'un bucket.
## L'idée derrière tout ca
Pour commencer, il y a la possibilité d'installer plus qu'un bucket sur l'api. Dans la pluspart des cas, on veut pouvoir rapidement acceder à un bucket et lire les informations des capteurs. C'est pourquoi les buckets seront afficher directement sur l'index du site, sous forme d'une liste. En sélectionnant un bucket, on arrive sur la liste des capteurs et on accède directement au informations que les senseurs nous envoie.

Ensuite, les fonctionnalitées diverses doivent être trouvable facilement. J'ai donc essayé de placer le tous à des endroits qui semblent logique. Le plus important était d'acceder au buckets rapidement, c'est donc pour cette raison qu'on doit accèder aux configurations de senseurs par la barre de navigation. Si vous avez des suggestions, n'hésiter pas.

AXDOOMER, 2018-06-22:
### Configuration de MQTTBox pour faire des tests

[image]
Configuration de MQTTBox. L'important c'est de mettre le bon protocole ( mqtt / tcp ) et le bon host (quand mosquitto est sur ton ordi: localhost:1883)