ClubCedille/jardiniot #49: [ARDUINO] Faire fonctionner les DELs

AXDOOMER, 2017-06-22:
On n'a pas le code source pour les DELs, il faut donc trouver leur fonctionnement. Elles doivent être contrôlées à partir des commandes reçues par MQTT. 

[image]

https://github.com/kiwisincebirth/Arduino/blob/master/libraries/PWMFrequency/PWMFrequency.h
http://playground.arduino.cc/Code/PwmFrequency
http://playground.arduino.cc/Main/TimerPWMCheatsheet
https://www.arduino.cc/en/Tutorial/SecretsOfArduinoPWM
https://github.com/ClubCedille/jardiniot/issues/21#issuecomment-294951640
https://www.arduino.cc/en/Tutorial/PWM

AXDOOMER, 2017-06-23:
Ça fonctionne avec le code suivant:
```
void setup() {
  // DELs -- 0 => fermé, 1 (faible intensité) à 170 (très fortes)
  // -- Les pins 9, 10 et 11 sotn des pines PWM
  analogWrite(9, 1);   // blanches
  analogWrite(10, 1);  // bleues
  analogWrite(11, 1);  // rouges
  
  // VENTILATEURS -- 0-127 => fermé, 128-255 => tourne
  analogWrite(8, 255);  // fan sur le heatsink. 
  analogWrite(7, 255);  // fan d'ordi sur le coté du bucket
}

void loop() {
  // rien
}
```

AXDOOMER, 2017-06-23:
J'ai plugué les fans dans des pins [PWM](https://www.arduino.cc/en/Tutorial/PWM), mais si la valeur n'est pas exactement 255 pour la faire tourner, on peut entendre du bruit électronique dedans, ce qui est très dérangeant. 

```
void setup() {
  // Les pins 5, 6, 9, 10, 11 reçoivent des PWM
  
  // DELs -- les valeurs données règlent l'intensité
  analogWrite(9, 1);   // blanches
  analogWrite(10, 1);  // bleues
  analogWrite(11, 1);  // rouges
  
  // VENTILATEURS -- la valeurs données règlent leur vitesse
  analogWrite(6, 255);  // fan sur le heatsink. 
  analogWrite(5, 255);  // fan d'ordi sur le coté du bucket
}

void loop() {
  // rien
}
```

AXDOOMER, 2017-06-23:
@cmccauley T'as fait une belle job avec l'électronique. J'ai pas eu de la misère du tout à faire fonctionner ça. 

Je passe maintenant à l'intégration de mon code dans le programme principal.

AXDOOMER, 2017-06-27:
Nouvelle configuration: 

[image]

Le ESP est connecté comme d'habitude: 

[image]

En général, ça l'air de ça:

[image]

Les senseurs et les DELs sont tous connectés ici pour le data: 

[image]

De l'autre côté ça reste pareil:

[image]

Faut être certain que le ventilateur latéral et le grand ventilateur soient bien connectés. (voir: le label reset qui fait face à la caméra et le fil jaune du grand ventilateur qui n'est connectés à aucune pin, car il y a juste 2 pins pour lui aussi comme l'autre). 

[image]

*Il n'y a aucun capaciteur pour le grand ventilateur, donc si on le fait pas tourner à 100% de sa vitesse, les PWM vont causer du bruit électronique dans l'air qu'on pourra entendre.

mikefaille, 2017-06-29:
Intéressant!

AXDOOMER, 2017-06-30:
En attente de correction d'un bug avant de tester officiellement le https://github.com/ClubCedille/jardiniot/pull/50 et de fermer ce issue.

AXDOOMER, 2017-07-07:
Comme le code a été testé dans un programme de test indépendant et qu'il marchait, je ferme ce issue.

closed 2017-07-07 by AXDOOMER