ClubCedille/jardiniot #55: Réorganisation du cablage du pot intelligent

AlexBrochu, 2017-11-27:
On devrait protéger les cables et les soudures dans une boîte. Ça permettrait de rendre le pot plus fonctionnelle et plus beau.

AXDOOMER, 2017-11-27:
Ça se ferait bien avec une imprimante 3D, comme la boîte qu'on avait pour cafETSière/nETSpresso. C'est quelque chose de quand même bien important de protéger les câbles et les soudures. La dernière fois que j'ai vu quelqu'un tirer la board jusqu'au bout des fils, j'ai failli faire une crise cardiaque. En plus, c'était arrivé juste après que j'avais réparé un fil qui avait lâché et que plus rien de fonctionnait. Ça serait dommage que ça short et qu'on soit obligé de tout refaire la board.

AXDOOMER, 2017-11-27:
Avant de procéder à faire une boîte, faudrait couper la board. Il y a une partie qu'on n'utilise pas et qu'on pourra utiliser pour faire un autre projet. Il faudra d'abord se mettre tous d'accord qu'on n'aura plus besoin d'ajouter des features sur la board.

notarock, 2018-03-20:
Es-ce que c'est related avec ce que notre ami d'ELE fait? Je crois qu'il voulait re-organisé le cablage.
Une fois que c'est fait, j'aimerai bien designé quelque choses qu'on pourrait faire imprimé en 3D

AXDOOMER, 2018-03-21:
Une fois qu'on aura un plan, on pourra se faire une belle board avec tout bien soudé dessus. On imprimera un case avec un imprimante 3D. Ça va être super propre et nice.

AXDOOMER, 2018-03-21:
Dépend de https://github.com/ClubCedille/jardiniot/issues/41