ClubCedille/jardiniot #56: Refactoriser le code du ESP pour le découper en classe CPP

AlexBrochu, 2017-12-07:
Trouver une solution pour éviter de mettre le mot de passe et les connexions du wifi directement dans le code du ESP. (Fichier externe sur le Arduino par exemple. (à voir si c'est possible))

AXDOOMER, 2017-12-08:
Ça se fait pas de mettre un fichier un fichier sur le Arduino ou sur le ESP. On est obligé de mettre les données du programme dans le code. C'est pour ça que j'ai fait un WiFi manager. 

Quand tu te connectes, le ESP devient un access point (pour le nombres de secondes défini dans le code) et tu peux te connecter pour entrer ton SSID de WiFi, ton mot de passe et l'IP du serveur. 

Par contre, je crois que le ESP va réutiliser le même WiFi si tu passes par le WiFi manager, et ce, jusqu'à ce qu'il soit reseté. C'est ce que tu cherches sûrement quand tu dis:

> Il faut aussi trouver une solution pour éviter de mettre le mot de passe et les connexions du wifi directement dans le code du ESP

Le WiFi manager c'est la solution à ne pas mettre de credentials dans le code. À moins qu'il existe un autre solution, j'en ai pas trouvé quand j'ai fais mes recherches. Le WiFi manager était la seule solution.

AXDOOMER, 2018-01-05:
> Refactoriser le code du ESP pour le découper en classe CPP

Est-ce qu'on peut maintenant dire que cela est accomplie ou reste-t-il du travail à faire?

Pull request en question: https://github.com/ClubCedille/jardiniot/pull/57

AXDOOMER, 2018-06-22:
Une chose que j'aimerais faire est groupe tous les .cpp et les .h dans le même ficher pour réduire la complexité des `#include`.

AXDOOMER, 2018-07-05:
Tous les bugs qui empêchaient le fonctionnement correcte du logiciel sur le Arduino ont été corrigés. Le tout est fonctionnel.

closed 2018-07-05 by AXDOOMER