ClubCedille/jardiniot #84: Intégration de Travis-CI pour tests de compilation automatisés

AXDOOMER, 2020-04-16:
Configurer des builds automatiques pour: 
* Le système embarqué :
  * être capable de compiler la ROM pour Arduino
  * être capable de compiler la ROM pour le ESP32
* Le serveur web
* Le front-end web (**???**)

**???** : ça peut-être que je le ferai pas. 

Youtube c'est toujours une bonne source d'information pour trouver comment configurer quelque chose.

AXDOOMER, 2020-05-06:
@notarock Si ça te tente, on peut faire ça ensemble.

notarock, 2020-05-06:
Fds prochaine?
cc @Perceval62

Perceval62, 2020-05-06:
Ouais absolument !

Perceval62, 2020-05-06:
https://github.com/mireq/esp32-simulator 
voici une image que l'on pourrait utiliser

Perceval62, 2020-05-06:
Pour clarifier il faudrait faire 2 pipelines, 1 pour la rev1 et un autre pour la rev 2

AXDOOMER, 2020-05-06:
Pour rev1, I guess on peut juste utiliser un image Ubuntu et installer `platformio` dedans pour build, ça devrait être good.

Perceval62, 2020-05-06:
Ouais, le lien que j'ai envoyé c'est pour la rev2