ClubCedille/jardiniot #89: Convention de code

Perceval62, 2020-05-13:
Il faut éditer le clang-format pour qu'il soit conforme aux critères de code de @AXDOOMER

AXDOOMER, 2020-05-13:
Le code style doit être corrigé dans https://github.com/ClubCedille/jardiniot/tree/hardware

@AXDOOMER  est responsable de créer un guide de convention de code.

AXDOOMER, 2020-05-13:
@notarock 

Pour le serveur web, tu peux suivre le PEP8 de Python. T'es libre de garder les tabs ou de changer pour les espaces. Si tu charges pour les espaces, ça va détruire le git-blame par contre. 

Pour le site web, fait ce que tu veux, c'est toi l'expert. 

Pour le code embarqué en C/C++, ça s'en vient.

AXDOOMER, 2020-09-02:
Voir https://github.com/ClubCedille/jardiniot/tree/master/jardiniot-emb#formatage-de-code pour la commande de formatage.